jiangcuo/Proxmox-Arm64 #1: (documentation) usage repack.sh

thefiredragon, 2022-12-21:
Hey, could you explain how to use your repack.sh, it looks like you only repack all x86 / amd packages to arm64 ones instead of building it as arm64 binarys from source code.

So proxmox only use scripts instead of binarys? 
Best regards

jiangcuo, 2022-12-22:
Hi , repack.sh only for  debian-crates-packages.These packages contains the source for the rust xxx crate,no binarys.
When repacked done, we can easy to slove proxmox-rust-packages depends  (E.g.  proxmox-apt,proxmox-backup-qemu) by using command ```mk-build-deps --install```. debian `devscripts` will reading `debian/control` and autoinstall depends.
cargo will search crates in `/usr/share/cargo/registry/` but not crates.io .
If you want know  how to build a normal promxox package in detail ,you can see my topic: https://foxi.buduanwang.vip/virtualization/pve/2000.html/

thefiredragon, 2022-12-22:
Thank you for your fast response,
So I have two questions, 
Will you maintain this project in future ?
We like to buy two arm servers for our company so this would be important to know.
Nice would be if you could provide this documentation also here on github to build proxmox for arm.

Second question could we automate this build process to stay easy up to date with proxmox

We would also donate for this project,
Best regards
Dan

jiangcuo, 2022-12-23:
Dear Dan
This project will always be maintained and kept in sync with the proxmox official version.
It's always a test project . If you are going to use this project, you need to be prepared to face problems.
Of course, I'll read each issue and try to fix it as best I can. 
In the next step, I will add some project information, installation manual ,build docs and notes to the readme.
Automatic build is not supported yet.The update speed is very fast now because the development environment has been prepared.Maybe a day or two later than the official release.

thefiredragon, 2022-12-26:
So your Proxmox Build works well,

Could you translate
https://github.com/jiangcuo/Proxmox-Arm64/blob/main/repacked.md

I had see, that you edited the readme here, perhaps a better place would be the wiki for this.
I'd like to compare the readme with the link you attached here;

https://foxi.buduanwang.vip/virtualization/pve/2000.html/

Nice would be to make all steps clear, at your blog you commented which packages need to build with aarch64 flag.

If you have a donation option or PayPal we would also donate for this great project
Very thanks to you.

closed 2023-04-22 by jiangcuo