jiangcuo/Proxmox-Arm64 #3: No Valid OpenPGP Data found

terrorpup, 2023-02-20:
Hi, 

So I am trying to use your version, but it will not get pass this on the Orange PI 5 

root@traveller:~# echo "deb https://mirrors.apqa.cn/proxmox/ pvearm  main">/etc/apt/sources.list.d/foxi.list
root@traveller:~# curl -L  https://mirrors.apqa.cn/proxmox/gpg.key |apt-key add
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
100  3667    0  3667    0     0    639      0 --:--:--  0:00:05 --:--:--   836
gpg: no valid OpenPGP data found.

root@traveller:~# apt update
Hit:1 http://armbian.tnahosting.net/apt bullseye InRelease
Hit:2 http://deb.debian.org/debian bullseye InRelease
Hit:3 http://security.debian.org bullseye-security InRelease
Hit:4 http://deb.debian.org/debian bullseye-updates InRelease
Hit:5 http://deb.debian.org/debian bullseye-backports InRelease
Get:6 https://mirrors.apqa.cn/proxmox pvearm InRelease [4,306 B]
Err:6 https://mirrors.apqa.cn/proxmox pvearm InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY B859507D6B1F46D3
Reading package lists... Done
W: GPG error: https://mirrors.apqa.cn/proxmox pvearm InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY B859507D6B1F46D3
E: The repository 'https://mirrors.apqa.cn/proxmox pvearm InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

jiangcuo, 2023-02-22:
Thanks, fix it!
Alist download url is `https://url/d/path/to/file`
Add curl and wget usr_agent for direct  download
```
		if ($http_user_agent ~* "(Ubuntu APT|Debian APT|curl|Wget)"){
			proxy_pass http://127.0.0.1:8001;
		}

```

closed 2023-02-22 by jiangcuo

xifi-kif, 2023-03-06:
@terrorpup did you manage to have this working on Orange PI 5? Is everything working? I'm interested, since my OPI5 is arriving.