jiangcuo/Proxmox-Arm64 #5: One CPU failed to come online in Raspi 4

lemniskett, 2023-03-17:
Hi, I've been using this for awhile and just realized that 1 CPU is missing.
[image]

```
root@lemnicloud:~# cat /proc/cpuinfo 
processor	: 0
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3
```

I saw this output in `dmesg`:
```
[    7.132666] CPU1: failed to come online
[    7.136495] CPU1: failed in unknown state : 0x0
[    7.141161] psci_checker: Error occurred (-5) while trying to power up CPU 1
```

I tried swapping operating system to Raspberry Pi OS to see if it's a hardware failure, but it seems to be fine in Raspberry Pi OS.

jiangcuo, 2023-03-17:
Hi.
I think it may be the cause of the kernel.
Please try to change to a kernel suitable for Raspberry Pi.
```
#if your pveroot=/dev/sda1,boot to RP system
#mount pveroot
mount /dev/sda1 /media
#copy modules
cp -r /lib/modules/`uname -r` /media/lib/modules/
#copy kernel, 
cp -r /boot/vmlinuz-`uname -r`  /media/boot/
#chroot
chroot /media update-initramfs -k all -u
#update-grub
mount -o bind /sys /media/sys
mount -o bind /proc /media/proc
mount -o bind /dev /media/dev
chroot /media
update-grub
```
Or give me your kernel config file, I can help you recompile the kernel .

lemniskett, 2023-03-20:
Hi, installing the 6.x edge kernel seems to fix it.
```
apt install pve-kernel-6.1-edge
```

previously I had 5.10.x as the kernel version.
Thank you for maintaining the repo!

closed 2023-03-20 by lemniskett