jiangcuo/Proxmox-Arm64 #6: Kunpeng 920arm page very stuck

Up-And-Coming-Lab, 2023-03-18:
After installing the Kunpeng 920arm server by following the tutorial, I found that when I used PVE web to create a VM, the VM installation guide page was very stuck

jiangcuo, 2023-03-18:
Can you share your vm config(`qm config {vmid}`) .
You can record  the vm installation  process if it is possible.

Up-And-Coming-Lab, 2023-03-20:
[image]

Up-And-Coming-Lab, 2023-03-20:
> Can you share your vm config(`qm config {vmid}`) . You can record the vm installation process if it is possible.

[image]

jiangcuo, 2023-03-20:
can you try another distro？ 
```https://rockylinux.org/download/```

closed 2023-04-22 by jiangcuo