jiangcuo/Proxmox-Arm64 #8: Installation Error

Angu0083, 2023-03-18:
I receiving an error while installing.

Backing up lvm.conf before setting pve-manager specific settings..
'/etc/lvm/lvm.conf' -> '/etc/lvm/lvm.conf.bak'
Setting 'global_filter' in /etc/lvm/lvm.conf to prevent zvols from being scanned:
global_filter=["a|.*|"] => global_filter=["r|/dev/zd.*|"]
Job for pvestatd.service failed because the control process exited with error code.
See "systemctl status pvestatd.service" and "journalctl -xe" for details.
dpkg: error processing package pve-manager (--configure):
 installed pve-manager package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of proxmox-ve:
 proxmox-ve depends on pve-manager; however:
  Package pve-manager is not configured yet.

dpkg: error processing package proxmox-ve (--configure):
 dependency problems - leaving unconfigured
Processing triggers for initramfs-tools (0.140) ...
update-initramfs: Generating /boot/initrd.img-5.10.0-21-arm64
Processing triggers for libglib2.0-0:arm64 (2.66.8-1) ...
Processing triggers for libc-bin (2.31-13+deb11u5) ...
Processing triggers for rsyslog (8.2102.0-2+deb11u1) ...
Processing triggers for ufw (0.36-7.1) ...
Processing triggers for man-db (2.9.4-2) ...
Setting up glib-networking:arm64 (2.66.0-2) ...
Setting up libsoup2.4-1:arm64 (2.72.0-2) ...
Setting up gstreamer1.0-plugins-good:arm64 (1.18.4-2+deb11u1) ...
Processing triggers for proxmox-backup-file-restore (2.3.3-1) ...
Updating file-restore initramfs...
12253 blocks
Processing triggers for libgdk-pixbuf-2.0-0:arm64 (2.42.2+dfsg-1+deb11u1) ...
Processing triggers for libc-bin (2.31-13+deb11u5) ...
Errors were encountered while processing:
 pve-manager
 proxmox-ve
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@proxmox:~# systemctl status pvestatd.service
● pvestatd.service - PVE Status Daemon
     Loaded: loaded (/lib/systemd/system/pvestatd.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sat 2023-03-18 22:30:44 UTC; 35s ago
    Process: 13572 ExecStart=/usr/bin/pvestatd start (code=exited, status=111)
        CPU: 463ms

Mar 18 22:30:44 proxmox pvestatd[13572]: ipcc_send_rec[1] failed: Connection refused
Mar 18 22:30:44 proxmox pvestatd[13572]: ipcc_send_rec[2] failed: Connection refused
Mar 18 22:30:44 proxmox pvestatd[13572]: ipcc_send_rec[1] failed: Connection refused
Mar 18 22:30:44 proxmox pvestatd[13572]: ipcc_send_rec[3] failed: Connection refused
Mar 18 22:30:44 proxmox pvestatd[13572]: Unable to load access control list: Connection refused
Mar 18 22:30:44 proxmox pvestatd[13572]: ipcc_send_rec[2] failed: Connection refused
Mar 18 22:30:44 proxmox pvestatd[13572]: ipcc_send_rec[3] failed: Connection refused
Mar 18 22:30:44 proxmox systemd[1]: pvestatd.service: Control process exited, code=exited, status=111/n/a
Mar 18 22:30:44 proxmox systemd[1]: pvestatd.service: Failed with result 'exit-code'.
Mar 18 22:30:44 proxmox systemd[1]: Failed to start PVE Status Daemon.
root@proxmox:~# journalctl -xe
Mar 18 22:31:29 proxmox pveproxy[22580]: worker exit
Mar 18 22:31:29 proxmox pveproxy[22582]: /etc/pve/local/pve-ssl.key: failed to load local private key (key_file or key) at /usr/share/perl5/PVE/APIServer/AnyEvent.pm line 1996.
Mar 18 22:31:29 proxmox pveproxy[22581]: worker exit
Mar 18 22:31:29 proxmox pveproxy[13553]: worker 22580 finished
Mar 18 22:31:29 proxmox pveproxy[13553]: starting 1 worker(s)
Mar 18 22:31:29 proxmox pveproxy[13553]: worker 22583 started
Mar 18 22:31:29 proxmox pveproxy[13553]: worker 22581 finished
Mar 18 22:31:29 proxmox pveproxy[13553]: starting 1 worker(s)
Mar 18 22:31:29 proxmox pveproxy[13553]: worker 22584 started
Mar 18 22:31:29 proxmox pveproxy[22583]: /etc/pve/local/pve-ssl.key: failed to load local private key (key_file or key) at /usr/share/perl5/PVE/APIServer/AnyEvent.pm line 1996.
Mar 18 22:31:29 proxmox pveproxy[22584]: /etc/pve/local/pve-ssl.key: failed to load local private key (key_file or key) at /usr/share/perl5/PVE/APIServer/AnyEvent.pm line 1996.
Mar 18 22:31:34 proxmox pveproxy[22582]: worker exit
Mar 18 22:31:34 proxmox pveproxy[13553]: worker 22582 finished
Mar 18 22:31:34 proxmox pveproxy[13553]: starting 1 worker(s)
Mar 18 22:31:34 proxmox pveproxy[13553]: worker 22585 started
Mar 18 22:31:34 proxmox pveproxy[22583]: worker exit
Mar 18 22:31:34 proxmox pveproxy[22585]: /etc/pve/local/pve-ssl.key: failed to load local private key (key_file or key) at /usr/share/perl5/PVE/APIServer/AnyEvent.pm line 1996.
Mar 18 22:31:34 proxmox pveproxy[22584]: worker exit
Mar 18 22:31:34 proxmox pveproxy[13553]: worker 22583 finished
Mar 18 22:31:34 proxmox pveproxy[13553]: starting 1 worker(s)
Mar 18 22:31:34 proxmox pveproxy[13553]: worker 22586 started
Mar 18 22:31:34 proxmox pveproxy[13553]: worker 22584 finished
Mar 18 22:31:34 proxmox pveproxy[13553]: starting 1 worker(s)
Mar 18 22:31:34 proxmox pveproxy[13553]: worker 22587 started
Mar 18 22:31:34 proxmox pveproxy[22586]: /etc/pve/local/pve-ssl.key: failed to load local private key (key_file or key) at /usr/share/perl5/PVE/APIServer/AnyEvent.pm line 1996.
Mar 18 22:31:34 proxmox pveproxy[22587]: /etc/pve/local/pve-ssl.key: failed to load local private key (key_file or key) at /usr/share/perl5/PVE/APIServer/AnyEvent.pm line 1996.
Mar 18 22:31:39 proxmox pveproxy[22585]: worker exit
Mar 18 22:31:39 proxmox pveproxy[13553]: worker 22585 finished
Mar 18 22:31:39 proxmox pveproxy[13553]: starting 1 worker(s)
Mar 18 22:31:39 proxmox pveproxy[13553]: worker 22588 started
Mar 18 22:31:39 proxmox pveproxy[22586]: worker exit
Mar 18 22:31:39 proxmox pveproxy[22588]: /etc/pve/local/pve-ssl.key: failed to load local private key (key_file or key) at /usr/share/perl5/PVE/APIServer/AnyEvent.pm line 1996.
Mar 18 22:31:39 proxmox pveproxy[22587]: worker exit
Mar 18 22:31:39 proxmox pveproxy[13553]: worker 22586 finished
Mar 18 22:31:39 proxmox pveproxy[13553]: starting 1 worker(s)
Mar 18 22:31:39 proxmox pveproxy[13553]: worker 22589 started
Mar 18 22:31:39 proxmox pveproxy[13553]: worker 22587 finished
Mar 18 22:31:39 proxmox pveproxy[13553]: starting 1 worker(s)
Mar 18 22:31:39 proxmox pveproxy[13553]: worker 22590 started
Mar 18 22:31:39 proxmox pveproxy[22589]: /etc/pve/local/pve-ssl.key: failed to load local private key (key_file or key) at /usr/share/perl5/PVE/APIServer/AnyEvent.pm line 1996.
Mar 18 22:31:39 proxmox pveproxy[22590]: /etc/pve/local/pve-ssl.key: failed to load local private key (key_file or key) at /usr/share/perl5/PVE/APIServer/AnyEvent.pm line 1996.

jiangcuo, 2023-03-19:
Hi! 
Did you add your ip and fqdn to your /etc/hosts ?
>
>For instance, if your IP address is 192.168.15.77, and your hostname prox4m1, then your /etc/hosts file could look like:
>```
>127.0.0.1       localhost.localdomain localhost
>192.168.15.77   prox4m1.proxmox.com prox4m1
>
># The following lines are desirable for IPv6 capable hosts
>::1     localhost ip6-localhost ip6-loopback
>ff02::1 ip6-allnodes
>ff02::2 ip6-allrouters
>```
>You can test if your setup is ok using the hostname command:
>
>```hostname --ip-address```
>192.168.15.77 # should return your IP address here
>
># ipcc_send_rec[1] failed
>
>ipcc_send_rec[1] failed: Connection refused
>
>then you should review your /etc/hosts file according to the instructions above.

https://github.com/jiangcuo/Proxmox-Arm64/wiki/Install-Proxmox-VE-on-Debian-bullseye

Angu0083, 2023-03-19:
Thanks, Issue got fixed.

closed 2023-03-19 by Angu0083