jiangcuo/Proxmox-Arm64 #12: [Install] packages have unmet dependencies:  pve-kernel-5.15 : Depends: pve-kernel-5.15.39-4-pve but it is not installable

mindingulove, 2023-03-25:
When io follow the  [Guide](https://github.com/jiangcuo/Proxmox-Arm64/wiki/Install-Proxmox-VE-on-Debian-bullseye) in my bullseye install, i get the following error: The following packages have unmet dependencies:
 pve-kernel-5.15 : Depends: pve-kernel-5.15.39-4-pve but it is not installable.

It seems that pve kernel is not found and install halts. What im i doing wrong? Cant seem to find a similar error neither an description about it. Thank you

jiangcuo, 2023-03-25:
Hello, thank you for discovering this question.
This is a dependency issue of the proxmox-ve package, I have now removed pve-kernel-5.15 from the proxmox-ve depend list, you can use `apt update ` and try again.

mindingulove, 2023-03-25:
> Hello, thank you for discovering this question.
> This is a dependency issue of the proxmox-ve package, I have now removed pve-kernel-5.15 from the proxmox-ve depend list, you can use `apt update ` and try again.

Now it worked, but at the end it faills to create a symbolic link as follows: 

`ln: failed to create symbolic link '/boot/pve/vmlinuz': Operation not permitted
ln: failed to create symbolic link '/boot/pve/initrd.img': Operation not permitted`

jiangcuo, 2023-03-26:
This is because your boot partition does not support soft links.
Just ignore it.

closed 2023-04-22 by jiangcuo