jiangcuo/Proxmox-Arm64 #16: SSH-Key validation

peio42, 2023-04-14:
Hello,

When trying to create a new CT giving a ssh public key, the process blocks in the end when the key is being validated (error 500)
I don't know if all arm64 systems are touched by this problem, but the script /usr/share/perl5/PVE/Tools.pm tries to create a temporary file with the key to send to 'ssh-keygen' for validation. To do so, it opens a file with O_TMPFILE flag, wich is hardcoded inside the script. But it looks like the value is different on arm64, and so file opening ends with "Invalid argument" error.
Maybe the script should be modified to use system-dependent value instead of hardcoding it, but a quick patch could be :

```
@@ -100,7 +100,7 @@

 use constant {O_PATH    => 0x00200000,
               O_CLOEXEC => 0x00080000,
-              O_TMPFILE => 0x00410000}; # This includes O_DIRECTORY
+              O_TMPFILE => 0x00404000}; # This includes O_DIRECTORY

 use constant {AT_EMPTY_PATH => 0x1000,
               AT_FDCWD => -100};
```

peio42, 2023-04-14:
As a sidenote, it looks like O_TMPFILE isn't (correctly ?) defined in perl Fnctl library, and must be the reason it was hardcoded. It's value is composed of the flag itself (0x00400000) and O_DIRECTORY which is the flag that differs on arm64 (0x10000 on amd64 while 0x4000 on arm64).
So maybe an other solution could be to declare

```
--- /usr/share/perl5/PVE/Tools.pm.orig	2023-04-14 11:29:28.388000000 +0200
+++ /usr/share/perl5/PVE/Tools.pm	2023-04-14 15:26:47.736000000 +0200
@@ -100,7 +100,7 @@

 use constant {O_PATH    => 0x00200000,
               O_CLOEXEC => 0x00080000,
-              O_TMPFILE => 0x00410000}; # This includes O_DIRECTORY
+              O_TMPFILE => 0x00400000 | O_DIRECTORY};

 use constant {AT_EMPTY_PATH => 0x1000,
               AT_FDCWD => -100};
```

which would work on most architectures.

jiangcuo, 2023-04-14:
Hi, thanks for your help. I tested both arm64 and amd64 platforms, and this patch works fine.I have committed the patch to this project.
I think you can commit  patch to proxmox too.

closed 2023-04-22 by jiangcuo