jiangcuo/Proxmox-Arm64 #17: Cannot get ip address from VM or CT on Orange Pi 5

lextor007, 2023-04-17:
Hi,

I have installed this on orange pi 5 everything seems well with some warning, but it’s running.

when deploying VM and container both unable to get ip address. I have try Linux bridge and ovs bridge same result.

Base os Armbian 23.02.2 Debian Bulleye
Pve-manager 7.4-3

Thanks.

jiangcuo, 2023-04-19:
Could you show:
- network config "/etc/network/interface"
- `ip a` command output
- vm config `qm config <vmid>` and `qm showcmd <vmid>`
- bridge show `brctl show`

lextor007, 2023-04-19:
Hi jiangcuo,

As below.

- Network config
source /etc/network/interfaces.d/*

auto lo
iface lo inet loopback

iface eth0 inet manual

iface enx00e04c681e8c inet manual

auto vmbr1
iface vmbr1 inet static
        address 192.168.0.1/24
        bridge-ports none
        bridge-stp off
        bridge-fd 0

auto vmbr0
iface vmbr0 inet manual
        bridge-ports none
        bridge-stp off
        bridge-fd 0

auto vmbr2
iface vmbr2 inet dhcp
        bridge-ports none
        bridge-stp off
        bridge-fd 0

auto vmbr3
iface vmbr3 inet manual
        ovs_type OVSBridge

- ip a
root@proxpi5:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 06:3a:dd:ec:db:4e brd ff:ff:ff:ff:ff:ff
    inet 10.0.3.16/22 brd 10.0.3.255 scope global dynamic noprefixroute eth0
       valid_lft 690961sec preferred_lft 690961sec
3: vmbr1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether 7e:83:c8:22:ea:1c brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.1/24 scope global vmbr1
       valid_lft forever preferred_lft forever
4: vmbr0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether 7e:79:d7:d4:9b:d4 brd ff:ff:ff:ff:ff:ff
5: vmbr2: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether 06:a3:31:9a:48:cc brd ff:ff:ff:ff:ff:ff
6: ovs-system: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether ea:d4:e9:c2:b8:f7 brd ff:ff:ff:ff:ff:ff
7: vmbr3: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UNKNOWN group default qlen 1000
    link/ether 3e:02:60:03:9c:41 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::3c02:60ff:fe03:9c41/64 scope link
       valid_lft forever preferred_lft forever

- vm config
root@proxpi5:~# qm config 100
bios: ovmf
boot: order=scsi0
cipassword: **********
ciuser: root
cores: 1
cpu: host
efidisk0: local:100/vm-100-disk-0.qcow2,efitype=4m,size=64M
ipconfig0: ip=dhcp
memory: 1024
meta: creation-qemu=7.2.0,ctime=1681662295
name: qemuvm
net0: virtio=BE:93:A5:FA:EC:7F,bridge=vmbr0
ostype: l26
scsi0: local:100/vm-100-disk-1.qcow2,size=2252M
scsi1: local:100/vm-100-cloudinit.qcow2,media=cdrom
scsihw: virtio-scsi-pci
serial0: socket
smbios1: uuid=872ce4d5-2b81-4fc5-a35a-b611dfc788dc

- qm showcmd
root@proxpi5:~# qm showcmd 100
/usr/bin/kvm -id 100 -name 'qemuvm,debug-threads=on' -no-shutdown -chardev 'socket,id=qmp,path=/var/run/qemu-server/100.qmp,server=on,wait=off' -mon 'chardev=qmp,mode=control' -chardev 'socket,id=qmp-event,path=/var/run/qmeventd.sock,reconnect=5' -mon 'chardev=qmp-event,mode=control' -pidfile /var/run/qemu-server/100.pid -daemonize -smbios 'type=1,uuid=872ce4d5-2b81-4fc5-a35a-b611dfc788dc' -drive 'if=pflash,unit=0,format=raw,readonly=on,file=/usr/share/pve-edk2-firmware//AAVMF_CODE.fd' -drive 'if=pflash,unit=1,id=drive-efidisk0,format=qcow2,file=/var/lib/vz/images/100/vm-100-disk-0.qcow2' -smp '1,sockets=1,cores=1,maxcpus=1' -nodefaults -boot 'menu=on,strict=on,reboot-timeout=1000,splash=/usr/share/qemu-server/bootsplash.jpg' -vnc 'unix:/var/run/qemu-server/100.vnc,password=on' -cpu host -m 1024 -device 'pci-bridge,id=pci.1,chassis_nr=1,bus=pcie.0,addr=0x1e' -device 'pci-bridge,id=pci.2,chassis_nr=2,bus=pcie.0,addr=0x1f' -readconfig /usr/share/qemu-server/pve-aarch64.cfg -device 'qemu-xhci,id=qemu-xchi' -device 'usb-tablet,id=tablet,bus=qemu-xchi.0,port=1' -device 'usb-kbd,id=keyboard,bus=qemu-xchi.0,port=2' -chardev 'socket,id=serial0,path=/var/run/qemu-server/100.serial0,server=on,wait=off' -serial chardev:serial0 -device 'virtio-gpu,id=vga,bus=pcie.0,addr=0x2' -device 'virtio-serial,id=spice,bus=pcie.0,addr=0x9' -chardev 'spicevmc,id=vdagent,name=vdagent' -device 'virtserialport,chardev=vdagent,name=com.redhat.spice.0' -spice 'tls-port=61000,addr=127.0.0.1,tls-ciphers=HIGH,seamless-migration=on' -device 'virtio-balloon-pci,id=balloon0,bus=pcie.0,addr=0x3,free-page-reporting=on' -iscsi 'initiator-name=iqn.1993-08.org.debian:01:fbac93b79735' -device 'virtio-scsi-pci,id=scsihw0,bus=pcie.0,addr=0x5' -drive 'file=/var/lib/vz/images/100/vm-100-disk-1.qcow2,if=none,id=drive-scsi0,format=qcow2,cache=none,aio=io_uring,detect-zeroes=on' -device 'scsi-hd,bus=scsihw0.0,channel=0,scsi-id=0,lun=0,drive=drive-scsi0,id=scsi0,bootindex=100' -drive 'file=/var/lib/vz/images/100/vm-100-cloudinit.qcow2,if=none,id=drive-scsi1,media=cdrom,aio=io_uring' -device 'scsi-cd,bus=scsihw0.0,channel=0,scsi-id=0,lun=1,drive=drive-scsi1,id=scsi1' -netdev 'type=tap,id=net0,ifname=tap100i0,script=/var/lib/qemu-server/pve-bridge,downscript=/var/lib/qemu-server/pve-bridgedown' -device 'virtio-net-pci,mac=BE:93:A5:FA:EC:7F,netdev=net0,bus=pcie.0,addr=0xb,id=net0,rx_queue_size=1024,tx_queue_size=1024' -machine 'type=virt+pve0,gic-version=host'

- brctl show
root@proxpi5:~# brctl show
bridge name     bridge id               STP enabled     interfaces
vmbr0           8000.7e79d7d49bd4       no              tap100i0
vmbr1           8000.7e83c822ea1c       no
vmbr2           8000.06a3319a48cc       no

Thanks for looking into it.

jiangcuo, 2023-04-20:
Your config is not right, follow this link to create:
https://pve.proxmox.com/wiki/Network_Configuration

lextor007, 2023-04-21:
Thanks for pointing it out, will look at it.

closed 2023-04-22 by jiangcuo