jiangcuo/Proxmox-Arm64 #19: Installing Proxmox VE from your repo fail

thegeek82000, 2023-04-23:
Hi,
I tried to install Proxmox on a SBC ROCK PI 5B :
I followed the WIKI and tried two times but it failed each time. 
rock@rock-5b:~$ uname -a
Linux rock-5b 5.10.66-27-rockchip-gea60d388902d #rockchip SMP Mon Oct 24 08:25:47 UTC 2022 aarch64 GNU/Linux
rock@rock-5b:~$ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye

The error is as follows :
 Preparing to unpack .../269-pve-firmware_3.6-4_all.deb ...
Unpacking pve-firmware (3.6-4) ...
dpkg: error processing archive /tmp/apt-dpkg-install-9ZsU3k/269-pve-firmware_3.6-4_all.deb (--unpack):
 trying to overwrite '/lib/firmware/ath10k/QCA988X/hw2.0/board.bin', which is also in package qualcomm-wifibt-firmware 0.1-1
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Selecting previously unselected package pve-manager.

Errors were encountered while processing:
 /tmp/apt-dpkg-install-9ZsU3k/269-pve-firmware_3.6-4_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

Any Help welcome

Thx

jiangcuo, 2023-04-24:
Hi, this is usually caused by conflicting firmware packages.
So you can remove the `qualcomm-wifibt-firmware`  and try again,or use `--force-overwrite`
```bash
apt download pve-firmware
dpkg -i --force-overwrite pve-firmware_*.deb
```

thegeek82000, 2023-05-20:
Hi,

The following command force install of six firmware package in conflict....
dpkg -i --force-overwrite pve-firmware_*.deb

Now Proxmox VE installed.

Just hope the wifi interface not broken.

Thx a lot.

closed 2023-06-18 by jiangcuo