jiangcuo/Proxmox-Arm64 #20: VM startup at 100 percent CPU and no console

lextor007, 2023-04-25:
Hi,

I have encounter issue on running VM guide from wiki:
`qm create 100 \
--name qemuvm \
--ostype l26 \
--cpu host \
--cores 1 \
--memory 1024 \
--scsihw virtio-scsi-pci \
--net0 virtio,bridge=vmbr0 \
--bios ovmf \
--efidisk0 local:0,efitype=4m,size=64M,format=qcow2 \
--serial0 socket \
--scsi0 local:0,import-from=/root/jammy-server-cloudimg-arm64.img,format=qcow2 \
--scsi1 local:cloudinit \
--ciuser root \
--cipassword rootroot \
--ipconfig0 ip=dhcp \
--boot order=scsi0 `

When I start the VM, the CPU goes 100 percent and never come down untill I stop the VM(waited for 30mins).
I open console for that VM shows "guess has not initialized the display (yet)."

This the qm config output:

`:~:# qm config 100
bios: ovmf
boot: order=scsi0
cipassword: **********
ciuser: root
cores: 2
cpu: max
efidisk0: local:100/vm-100-disk-0.qcow2,efitype=4m,size=64M
ipconfig0: ip=dhcp
memory: 1024
meta: creation-qemu=7.2.0,ctime=1682406393
name: qemuvm
net0: virtio=82:97:17:86:E3:06,bridge=vmbr0
numa: 0
ostype: l26
scsi0: local:100/vm-100-disk-1.qcow2,size=2252M
scsi1: local:100/vm-100-cloudinit.qcow2,media=cdrom
scsihw: virtio-scsi-pci
serial0: socket
smbios1: uuid=5c9db09c-05c4-4072-8d71-23cb32d4e8ee
sockets: 1
vga: virtio
`

Thanks in advance!

jiangcuo, 2023-04-25:
Hi, 
What hardware do you have?
Could you open the xtermjs console of the vm and  watch its output.

jarrodthebobo, 2023-04-25:
I also just started having this same issue out of nowhere after having my vms work for over 2 weeks at this point. Ive attempted to reinistall the entire os from scratch to no evail. Im assuming there is some sort of bug in one of the latest updates.

jiangcuo, 2023-04-26:
@jarrodthebobo @lextor007  
Hi,  can you show pveversion ` pveversion --verbos` and  `qm showcmd {vmid}` output.
For rk3389 rk3588, I suggest that set cpu affinity to use a55 or a76 only.

jiangcuo, 2023-04-26:
Temporary solution is to downgrade the pve-edk2-firmware package。
```bash
apt download  pve-edk2-firmware=3.20220526-1 
dpkg -i pve-edk2-firmware_3.20220526-1_all.deb
```
and reboot your system.

Version 3.20220526-1 has been remove from repo.

lextor007, 2023-04-26:
After downgrade, it works!

Thanks for the help!

jiangcuo, 2023-04-27:
report the bug:
https://bugzilla.proxmox.com/show_bug.cgi?id=4696

jiangcuo, 2023-04-28:
b5d34b95b8aaac4c80c9f39ee53eb86211db9bf3 fixed

closed 2023-04-28 by jiangcuo