jiangcuo/Proxmox-Arm64 #25: Pcie passthrough problem

tempus74, 2023-06-11:
Hi

I get this message when i try and do a pcie passthrough.

No IOMMU detected, please activate it.See Documentation for further information.

i have added this in grub: GRUB_CMDLINE_LINUX_DEFAULT="quiet iommu.passthrough=1"
and this in /etc/modules
vfio
vfio_iommu_type1
vfio_pci
vfio_virqfd

How can i get this working ?

jiangcuo, 2023-06-13:
What hardware are you using？

tempus74, 2023-06-16:
Im using a huawei taishan 2280 kunpeng 920

jiangcuo, 2023-06-17:
Make sure  smmu has been  enabled by bios

cronop, 2023-06-20:
Hi @jiangcuo did you compile your own kernel to get smmu support? on orange pi 5 plus, even though rk3588 has [MMU600](https://github.com/FanX-Tek/rk3588-TRM-and-Datasheet/blob/master/Rockchip%20RK3588%20TRM%20V1.0-Part1-20220309.pdf), it does not seem to be enabled in either armbian or orange debian.

jiangcuo, 2023-06-23:
@cronop  Symbol `ROCKCHIP_IOMMU=y`   has  been enabled in kernel default config. See [Rockchip IOMMU Support](https://www.kernelconfig.io/config_rockchip_iommu) 

I think that mybe no kernel's problem.  There has a iommu doc
https://www.kernel.org/doc/Documentation/devicetree/bindings/iommu/rockchip%2Ciommu.txt

jiangcuo, 2023-06-23:
@tempus74 kernel has enabled iommu by default . https://github.com/jiangcuo/pve-arm64-kernel/raw/master/debian/config/config.pve
You can check server's bios and make sure smmu has been enabled .

cronop, 2023-06-23:
Thank you @jiangcuo and @tempus74! I think this is down to rockchip/orangepi mmu support - https://www.reddit.com/r/OrangePI/comments/14g4cob/orange_pi_5_plus_rk3588_iommu_smmu_vfio_pci

tempus74, 2023-06-25:
Thx @jiangcuo

After looking closer i have a Taishan 2080 v1 server with hi1616 cpu that I'm reading as kunpeng 916 but I'm not sure, i have no settings for smmu or any settings regarding virtualization in bios 2080 V2 looks like it has it, I'm having trouble downloading new bios from Huawei since om not a company and got the server used from ebay so it's from 2017, so if anyone has enterprise access to Huawei hook me up 😀

Everything else is working good thx for maintaining this repo.
/Urban

jiangcuo, 2023-06-30:
@tempus74 Huawei docs are not pulic.
Try this path
“Advanced > MICC Config > Support Smmu”

closed 2023-08-05 by jiangcuo