jiangcuo/Proxmox-Arm64 #26: oragepi 5b

HolgerMohr, 2023-06-19:
I have installed Debian (from the manufacturer) on my oragepi 5b on a sd card and started the setup after the wiki it only aborts the installation at this point with 

Unpacking pve-firmware (3.6-4) ...
dpkg: error processing archive /tmp/apt-dpkg-install-7KrmaM/259-pve-firmware_3.6-4_all.deb (--unpack):
 trying to overwrite '/lib/firmware/brcm/brcmfmac4329-sdio.bin', which is also in package orangepi-firmware 1.0.2
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Selecting previously unselected package pve-manager.
Preparing to unpack .../260-pve-manager_7.4-3_arm64.deb ...
Unpacking pve-manager (7.4-3) ...
Selecting previously unselected package proxmox-ve.
Preparing to unpack .../261-proxmox-ve_7.3-1_all.deb ...
Unpacking proxmox-ve (7.3-1) ...
Errors were encountered while processing:
 /tmp/apt-dpkg-install-7KrmaM/259-pve-firmware_3.6-4_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@orangepi5b:~#

Translated with www.DeepL.com/Translator (free version)

jiangcuo, 2023-06-23:
https://github.com/jiangcuo/Proxmox-Arm64/issues/19

HolgerMohr, 2023-06-26:
hat sich erledigt

closed 2023-06-26 by HolgerMohr