jiangcuo/Proxmox-Arm64 #27: error instaling from repo smartmontools_7.2-pve1_arm64.deb  File has unexpected size (527124 != 535744)

Pull3D, 2023-06-25:
error instaling from repo 
smartmontools_7.2-pve1_arm64.deb  File has unexpected size (527124 != 535744)

`root@prox4m1:/home/pull# root@prox4m1:/home/pull# apt install proxmox-ve postfix open-iscsi
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
open-iscsi is already the newest version (2.1.3-5).
postfix is already the newest version (3.5.18-0+deb11u1).
proxmox-ve is already the newest version (7.3-1).
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 smartmontools : Depends: libc6 (>= 2.34) but 2.31-13+rpt2+rpi1+deb11u5 is to be installed
                 Depends: libstdc++6 (>= 11) but 10.2.1-6 is to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
root@prox4m1:/home/pull# apt --fix-broken install
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Correcting dependencies... Done
The following additional packages will be installed:
  smartmontools
Suggested packages:
  gsmartcontrol smart-notifier
The following packages will be upgraded:
  smartmontools
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
8 not fully installed or removed.
Need to get 536 kB of archives.
After this operation, 47.1 kB disk space will be freed.
Do you want to continue? [Y/n] Y
Get:1 https://mirrors.apqa.cn/proxmox pvearm/main arm64 smartmontools arm64 7.2-pve1 [536 kB]
Err:1 https://mirrors.apqa.cn/proxmox pvearm/main arm64 smartmontools arm64 7.2-pve1
  File has unexpected size (527124 != 535744). Mirror sync in progress? [IP: 152.70.92.238 443]
  Hashes of expected file:
   - SHA512:6c791ecc05fd659295d64b66638c7e18d11528adbbcfe8c65c75ca49c7499d024ff2b2587d2579fb717a5d3eec8d95b4665253a41989d80bfc3bff7de4d38121
   - SHA256:fb4a33e9124d1b1d8f7f3755fef5b70146d0d007916f793b6b3079a9bcef5a63
   - SHA1:7308a5b89b49b2cda07759d6ca53a9016136b649 [weak]
   - MD5Sum:89bef5b7d2de915d358f5405495cd514 [weak]
   - Filesize:535744 [weak]
E: Failed to fetch https://mirrors.apqa.cn/proxmox/pool/main/s/smartmontools/smartmontools_7.2-pve1_arm64.deb  File has unexpected size (527124 != 535744). Mirror sync in progress? [IP: 152.70.92.238 443]
   Hashes of expected file:
    - SHA512:6c791ecc05fd659295d64b66638c7e18d11528adbbcfe8c65c75ca49c7499d024ff2b2587d2579fb717a5d3eec8d95b4665253a41989d80bfc3bff7de4d38121
    - SHA256:fb4a33e9124d1b1d8f7f3755fef5b70146d0d007916f793b6b3079a9bcef5a63
    - SHA1:7308a5b89b49b2cda07759d6ca53a9016136b649 [weak]
    - MD5Sum:89bef5b7d2de915d358f5405495cd514 [weak]
    - Filesize:535744 [weak]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
`

3dn0, 2023-06-27:
Hello !

I've got the same error trying to install it on top of a debian11 on an ampere platform.
The workaround I've tried is working but I don't think that a Proxmox installed this way would be good for production.
Anyway.

Just grab the corresponding iso file and mount it. Then install the missing package from the package folder on the iso.
Re-start the proxmox install, and it should go smoothly.

Have fun testing :)

ja-finn, 2023-06-28:
Same issue here converting from a Pimox install. Instead of loading it from the ISO you can just install it from the main repo. You can either pin the package or, if you don't know how to do that, just comment out the proxmox-arm repo, run `apt update && apt install smartmontools` and then enable the proxmox-arm repo again.

jiangcuo, 2023-06-30:
Hi all. 
I'm very sorry for this.
This issue may cause by package break when sync repo.
My home server has failed and I'm on a business trip. I can't fix this issue right now.
The temporary workaround is to download the problematic package and then install it.
like  smartmontools
```bash
wget https://mirrors.apqa.cn/d/proxmox/pool/main/s/smartmontools/smartmontools_7.2-pve1_arm64.deb
dpkg -i smartmontools_7.2-pve1_arm64.deb
```

the-impaler, 2023-06-30:
I came here with this problem too, but dpkg throws a dependency problem when I try to force install the smartmontools package after using wget to grab it using the above instructions :-(

the dependency problems are libc6 and libstdc++6 which both say that smartmontools requires a higher level (> 2.34 abd >11 respectiely) than the version which is installed on the system currently (2.31-13 and 10.2.1-6 respectively).

Any ideas or shall we just be patient? :-)

Thanks!

jiangcuo, 2023-07-07:
Hi all, we can build smartmontools from source code.
for pve8
```bash
apt update && apt install build-* git  devscripts -y
git clone  https://git.proxmox.com/git/smartmontools.git
cd smartmontools
git checkout proxmox/bookworm
yes| mk-build-deps --install --remove
make deb
```
for pve7
```bash
apt update && apt install build-* git  devscripts -y
git clone  https://git.proxmox.com/git/smartmontools.git
cd smartmontools
git checkout proxmox/bullseye
yes| mk-build-deps --install --remove
make deb
```

jiangcuo, 2023-07-09:
fixed.

closed 2023-07-12 by jiangcuo