jiangcuo/Proxmox-Arm64 #28: VMs no longer boot with upgrade from 7.4 to 8

jarrodthebobo, 2023-06-29:
After following the documented steps to upgrade to the newest proxmox version, my vms now no longer boot. I also cannot get into the uefi menu. It appears that it just completly hangs.

It looks like the issue we had here https://github.com/jiangcuo/Proxmox-Arm64/issues/20 may have returned?

Im on the rock 5b

titogrima, 2023-06-29:
Hi!!

I have same error and I can solved it with downgrade pve-edk2-firmware same that #20 and create a symbolic link in this path /usr/share/pve-edk2-firmware/

[image]

In theory it was solved in the version of the package that is installed with the PVE 8 in this ticket 
https://bugzilla.proxmox.com/show_bug.cgi?id=4696  open by @jiangcuo but don't work

Im on orange pi 5

jarrodthebobo, 2023-06-29:
When I try to downgrade the package I'm given an error stating there is no download candidate/it is outdated or reference by another package. 

Not sure what I'm doing wrong here, haha

jiangcuo, 2023-06-30:
@jarrodthebobo Can you show your pveversion and vm config?
edk2 issue has been resloved by proxmox.

ruslanvd, 2023-06-30:
@jiangcuo 

Hello. With the new pve-edk2-firmware=3.20230228-4 nothing is solved either :(

titogrima, 2023-06-30:
Hi!

I installed pve-edk2-firmware_3.20230228-4_all.deb package download directly from Proxmox 8 repo and not work
wget http://download.proxmox.com/debian/pve/dists/bookworm/pve-no-subscription/binary-amd64/pve-edk2-firmware_3.20230228-4_all.deb
dpkg -i pve-edk2-firmware_3.20230228-4_all.deb

[image]

If downgrade again to pve-edk2-firmware_3.20220526-1_all.deb work without any problem again 

[image]

@jarrodthebobo try to downgrade to pve-edk2-firmware_3.20220526-1_all.deb package and create de symbolink link

Downgrade edk2
wget http://download.proxmox.com/debian/pve/dists/bullseye/pve-no-subscription/binary-amd64/pve-edk2-firmware_3.20220526-1_all.deb
 dpkg -i pve-edk2-firmware_3.20220526-1_all.deb

create symboly link
cd /usr/share/pve-edk2-firmware
 ln -s AAVMF_CODE.fd AAVMF_CODE.ms.fd

and execute this comand  to set gic-version=host and up de VM again
 I hope useful...

qm set vmid --args "-machine virt,gic-version=host"

in your case should be
qm set 105 --args "-machine virt,gic-version=host"

Regards

ruslanvd, 2023-06-30:
@titogrima Thank you so much :). A firmware downgrade helped. Old VMs also start normally.

titogrima, 2023-06-30:
good!!

And @jiangcuo thank you very much for this port and your work it's awesome for home labs ;) 

Regardas

jiangcuo, 2023-07-09:
I have tested the lastest pve-edk2-firmware version 3.20230228-4. It works.

I think this issus cause by my mistake .I missed the code about the gic version when I optimized the patch. 

Now，New version of qemu-server(8.0.6-1) has upload,You can update it.

titogrima, 2023-07-09:
In Orangepi5 with gic-version=host, pve-edk2-firmware version 3.20230228-4 and qemu-server 8.0.6 dont work and with qemu-server 8.0.6-1 neither work but work with pve-edk2-firmware 3.20220526-1 without problems
With Raspberry Pi 4 work with pve-edk2-firmware version 3.20230228-4 and qemu-server 8.0.6 without any problem 

Maybe this problem only ocurrs with RK3588 SOC 

Regards

itdivan, 2023-07-19:
**OrangePI 5 Plus** has the same issue 
100% CPU load and nothing shows in console
```
chip RK3588
Armbian Bookwarm (Debian 12)
5.10.160-rk35xx
```
installed from `deb https://mirrors.apqa.cn/proxmox/ pvearm8 main`
```
proxmox-ve: 8.0.1 (running kernel: 5.10.160-rk35xx)
pve-manager: 8.0.3 (running version: 8.0.3/6d727f81c8ca23b5)
ceph-fuse: 17.2.6-pve1+3
corosync: 3.1.7-pve3
criu: 3.17.1-2
glusterfs-client: 10.3-5
ifupdown: residual config
ifupdown2: 3.2.0-1+pmx2
libjs-extjs: 7.0.0-3
libknet1: 1.25-1
libproxmox-acme-perl: 1.4.6
libproxmox-backup-qemu0: 1.4.0
libproxmox-rs-perl: 0.3.0
libpve-access-control: 8.0.3
libpve-apiclient-perl: 3.3.1
libpve-common-perl: 8.0.5
libpve-guest-common-perl: 5.0.3
libpve-http-server-perl: 5.0.3
libpve-rs-perl: 0.8.3
libpve-storage-perl: 8.0.1
libspice-server1: 0.15.1-1
lvm2: 2.03.16-2
lxc-pve: 5.0.2-4
lxcfs: 5.0.3-pve3
novnc-pve: 1.4.0-2
proxmox-backup-client: 2.99.0-1
proxmox-backup-file-restore: 2.99.0-1
proxmox-kernel-helper: 8.0.2
proxmox-mail-forward: 0.2.0
proxmox-mini-journalreader: 1.4.0
proxmox-widget-toolkit: 4.0.5
pve-cluster: 8.0.1
pve-container: 5.0.3
pve-docs: 8.0.3
pve-edk2-firmware: 3.20230228-4
pve-firewall: 5.0.2
pve-ha-manager: 4.0.2
pve-i18n: 3.0.4
pve-qemu-kvm: 8.0.2-3
pve-xtermjs: 4.16.0-3
qemu-server: 8.0.6-1
smartmontools: 7.3-pve1
spiceterm: 3.3.0
swtpm: 0.8.0+pve1
vncterm: 1.8.0
```

via strace i got this one on promblem process id (pehaps it hepls)
```
write(30, "\344\361\364\270\377\377\0\0\0\0\0\0\2\0\0\0", 16) = 16
ppoll([{fd=3, events=POLLIN}, {fd=4, events=POLLIN}, {fd=7, events=POLLIN}, {fd=8, events=POLLIN}, {fd=9, events=POLLIN}, {fd=12, events=POLLIN}, {fd=13, events=POLLIN}, {fd=14, events=POLLIN}, {fd=16, events=POLLIN}, {fd=20, events=POLLIN}, {fd=28, events=POLLIN}], 11, {tv_sec=0, tv_nsec=0}, NULL, 8) = 0 (Timeout)
ppoll([{fd=3, events=POLLIN}, {fd=4, events=POLLIN}, {fd=7, events=POLLIN}, {fd=8, events=POLLIN}, {fd=9, events=POLLIN}, {fd=12, events=POLLIN}, {fd=13, events=POLLIN}, {fd=14, events=POLLIN}, {fd=16, events=POLLIN}, {fd=20, events=POLLIN}, {fd=28, events=POLLIN}], 11, {tv_sec=0, tv_nsec=29565212}, NULL, 8) = 0 (Timeout)
```

ps OrangePI5 and OrangePI5B has chip RK3588S

jiangcuo, 2023-07-19:
@itdivan 
Can you try downgrade edk2? 
```
wget http://download.proxmox.com/debian/pve/dists/bullseye/pve-no-subscription/binary-amd64/pve-edk2-firmware_3.20220526-1_all.deb
dpkg -i pve-edk2-firmware_3.20220526-1_all.deb
```

jiangcuo, 2023-07-19:
follow
https://github.com/jiangcuo/Proxmox-Arm64/issues/28#issuecomment-1614667068

itdivan, 2023-07-20:
> @itdivan Can you try downgrade edk2?
> 
> ```
> wget http://download.proxmox.com/debian/pve/dists/bullseye/pve-no-subscription/binary-amd64/pve-edk2-firmware_3.20220526-1_all.deb
> dpkg -i pve-edk2-firmware_3.20220526-1_all.deb
> ```

Yes, it works

jiangcuo, 2023-08-05:
1. updated correct pkg in repo
2. updated this issue on wiki

closed 2023-08-05 by jiangcuo