jiangcuo/Proxmox-Arm64 #33: Kernel config to support SolidRun LX2K platforms

ndoo, 2023-07-22:
Hi, I do not have working Ethernet on SolidRun LX2K (Freescale/NXP QoriQ Ethernet);

A good start would be if we could add the following kernel configs?
https://github.com/SolidRun/lx2160a_build/blob/master/configs/linux/lx2k_additions.config

Thank you!

jiangcuo, 2023-07-23:
HI. `https://github.com/jiangcuo/pve-port-kernel.git` This project may not be suitable for all machines.
I can create a branch based on your machine. But I don't have enough machines for this, so you'll need to compile it by yourself.

ndoo, 2023-07-23:
Sure, I am happy to test. I have your project working on my board, except
for the Freescale mc bus and ethernet.

jiangcuo, 2023-07-23:
Hi,You can try it.
https://github.com/jiangcuo/pve-port-kernel/tree/lsdk-21.08

ndoo, 2023-07-23:
Thank you, I tried to build twice, however:

```
make[2]: *** [Makefile:1821: drivers] Error 2
make[2]: Leaving directory '/root/pve-port-kernel/linux'
dh_auto_build: error: cd linux && make -j16 "INSTALL=install --strip-program=true" CC=cc KCFLAGS= EXTRAVERSION=-lsdk LOCALVERSION= "KBUILD_BUILD_VERSION_TIMESTAMP=PVE Edge 5.10.35 (2023-07-23)" returned exit code 2
make[1]: *** [debian/rules:60: override_dh_auto_build] Error 25
make[1]: Leaving directory '/root/pve-port-kernel'
make: *** [debian/rules:16: build] Error 2
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
debuild: fatal error at line 1182:
dpkg-buildpackage -us -uc -ui --jobs=auto -b failed
```

jiangcuo, 2023-08-05:
I see that some patches are not compatible with kernel. You can remove it in debian/patches/series.linux

closed 2023-08-28 by jiangcuo