jiangcuo/Proxmox-Arm64 #38: aarch64 support for Parallels on macOS Apple Silicon?

ylluminate, 2023-08-05:
Would be very useful to have Proxmox available on macOS Parallels with Apple Silicon so that we can run native speed LXC containers. Any thoughts on how this might be accomplished?

jiangcuo, 2023-08-05:
Proxmox-Arm64 can work on Apple Silicon with docker.
So there are two ways to support proxmox-arm64 on Apple Silicon.
1. Create  a debian vm and install proxmox-ve
2. Install debian on Apple Silicon and install proxmox-ve.

ylluminate, 2023-08-05:
Appreciate the recommendation. I really like the idea of running Debian 12 as the base via its `arm64` (aka properly `aarch64`) iso, but the ZFS setup is a little on the laborious side. It seems to me that your `arm64` can also work. Just threw it into Parallels and it detected and installed just fine (very, very fast). Unfortunately your download sources are not working for updates on 8:
```
root@mbp-pve:~# apt update
Ign:1 https://mirrors.apqa.cn/proxmox/debian/proxmox/pve bookworm InRelease          
Err:2 https://mirrors.apqa.cn/proxmox/debian/proxmox/pve bookworm Release            
  404  Not Found [IP: 152.70.92.238 443]
Ign:3 https://mirrors.ustc.edu.cn/debian bookworm InRelease
Ign:4 https://mirrors.ustc.edu.cn/debian bookworm-updates InRelease
Ign:5 https://mirrors.ustc.edu.cn/debian bookworm-backports InRelease
Ign:6 https://mirrors.ustc.edu.cn/debian-security bookworm-security InRelease
Ign:3 https://mirrors.ustc.edu.cn/debian bookworm InRelease
Ign:4 https://mirrors.ustc.edu.cn/debian bookworm-updates InRelease
Ign:5 https://mirrors.ustc.edu.cn/debian bookworm-backports InRelease
Ign:6 https://mirrors.ustc.edu.cn/debian-security bookworm-security InRelease
Ign:3 https://mirrors.ustc.edu.cn/debian bookworm InRelease
Ign:4 https://mirrors.ustc.edu.cn/debian bookworm-updates InRelease
Ign:5 https://mirrors.ustc.edu.cn/debian bookworm-backports InRelease
Ign:6 https://mirrors.ustc.edu.cn/debian-security bookworm-security InRelease
Err:3 https://mirrors.ustc.edu.cn/debian bookworm InRelease
  Cannot initiate the connection to mirrors.ustc.edu.cn:443 (2001:da8:d800:95::110). - connect (101: Network is unreachable) Could not connect to mirrors.ustc.edu.cn:443 (202.141.160.110), connection timed out Could not connect to mirrors.ustc.edu.cn:443 (202.141.176.110), connection timed out
Err:4 https://mirrors.ustc.edu.cn/debian bookworm-updates InRelease
  Cannot initiate the connection to mirrors.ustc.edu.cn:443 (2001:da8:d800:95::110). - connect (101: Network is unreachable)
Err:5 https://mirrors.ustc.edu.cn/debian bookworm-backports InRelease
  Cannot initiate the connection to mirrors.ustc.edu.cn:443 (2001:da8:d800:95::110). - connect (101: Network is unreachable)
Err:6 https://mirrors.ustc.edu.cn/debian-security bookworm-security InRelease
  Cannot initiate the connection to mirrors.ustc.edu.cn:443 (2001:da8:d800:95::110). - connect (101: Network is unreachable)
Reading package lists... Done
E: The repository 'https://mirrors.apqa.cn/proxmox/debian/proxmox/pve bookworm Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

jiangcuo, 2023-08-06:
Thanks. 
This is caused by the wrong proxmox-arm64 repo url. 
You can follow [wiki](https://github.com/jiangcuo/Proxmox-Port/blob/main/help/repo.md) and correct it.
9a66cc6d368385e11a0092a13303e7992e83ea95  has fixed it 

The iso default debian repo mirror is location on ustc.edu.cn. You can change it to debian default url .
```
deb http://deb.debian.org/debian bookworm main contrib
deb http://deb.debian.org/debian bookworm-updates main contrib

# security updates
deb http://security.debian.org/debian-security bookworm-security main contrib
```

ylluminate, 2023-08-06:
Nice, thank you very much. Please verify these look sane and proper:

```
$ cat /etc/apt/sources.list
deb http://deb.debian.org/debian bookworm main contrib non-free non-free-firmware
deb http://deb.debian.org/debian bookworm-updates main contrib non-free non-free-firmware
deb http://deb.debian.org/debian bookworm-backports main contrib non-free non-free-firmware

# security updates
deb http://security.debian.org/debian-security bookworm-security main contrib non-free non-free-firmware
```

```
$ cat /etc/apt/sources.list.d/pveport.list
deb https://global.mirrors.apqa.cn/proxmox/debian/pve bookworm port
```

Furthermore, can you verify that having these in place will perpetually update the PVE 8 system as you move forward past testing? I'd like to use this long term here if and until maybe the main Proxmox guys integrate your work. 

This is a fantastic effort. Thank you for your work!!! <3

jiangcuo, 2023-08-06:
Yes. This project will be kept up to date for a long time. 
[Proxmox-Port](https://github.com/jiangcuo/Proxmox-Port.git) is designed to port pve and pbs to different architectures. Arm64、RISC-V、Loongarch are supported now .

ylluminate, 2023-08-06:
Perhaps better on another thread, but just to throw this out here:

Do you have a mirror or method to install `arm64`/`aarch64` LXC images yet via `pveam`?

jiangcuo, 2023-08-06:
https://github.com/jiangcuo/Proxmox-Arm64/wiki/Linux-Container

ylluminate, 2023-08-07:
Thanks. Hmm, yes, I was hoping we could figure out a method that would facilitate them all being simply listed / integrated into the system for simpler installation / better experience. Hmmmmm.

closed 2023-08-28 by jiangcuo