jiangcuo/Proxmox-Arm64 #39: E: The repository 'https://mirrors.apqa.cn/proxmox pvearm Release' does not have a Release file.

mindingulove, 2023-08-06:
Followed wiki post and everytime i tried to update i get 
`root@prox4m1:/home/pi# apt update && apt full-upgrade
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian bookworm-updates InRelease                                                                                    
Hit:3 http://security.debian.org/debian-security bookworm-security InRelease                                                                     
Hit:4 http://archive.raspberrypi.org/debian bullseye InRelease                                                                                   
Hit:5 https://global.mirrors.apqa.cn/proxmox/debian/pve bullseye InRelease                   
Ign:6 https://mirrors.apqa.cn/proxmox pvearm InRelease                 
Err:7 https://mirrors.apqa.cn/proxmox pvearm Release
  404  Not Found [IP: 152.70.92.238 443]
Reading package lists... Done
E: The repository 'https://mirrors.apqa.cn/proxmox pvearm Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
N: Repository 'Debian bookworm' changed its 'non-free component' value from 'non-free' to 'non-free non-free-firmware'
N: More information about this can be found online in the Release notes at: https://www.debian.org/releases/bookworm/arm64/release-notes/ch-information.html#non-free-split`

Any idea of whats happening?
Thank you

closed 2023-08-06 by mindingulove

TheBossME, 2023-08-06:
IPv6 configuration

mindingulove, 2023-08-06:
The problem was that my dhcp was blocking https://mirrors.apqa.cn and screwing things up.