jiangcuo/Proxmox-Arm64 #40: error when start the vm

yuhongwei380, 2023-08-25:
my armpve is upgrade 7.3 to 7.4-16 ; when I start the vm, it noticed that 
EFI base image '/usr/share/pve-edk2-firmware//AAVMF_CODE.ms.fd' not found 
my vm setting 
[image]

yuhongwei380, 2023-08-25:
[image]

jiangcuo, 2023-08-25:
What is the version of pve-edk2-firmware？
Can you show this command output `ls /usr/share/pve-edk2-firmware/ -l`

yuhongwei380, 2023-08-25:
[image]

yuhongwei380, 2023-08-25:
> What is the version of pve-edk2-firmware？ Can you show this command output `ls /usr/share/pve-edk2-firmware/ -l`
OK
[image]

yuhongwei380, 2023-08-25:
the edk2-fireware 's version
[image]

yuhongwei380, 2023-08-25:
I can't find AAVMF_CODE.ms.fd;where i can get it!

yuhongwei380, 2023-08-25:
> AAVMF_CODE.ms.fd

I check X86 PVE 's file , the same file alse doesn't exist .

yuhongwei380, 2023-08-25:
I touch this file ,but  I met the new problem : when I start the vm 

generating cloud-init ISO
kvm: device requires 67108864 bytes, block backend provides 0 bytes
TASK ERROR: start failed: QEMU exited with code 1

albertogrimana, 2023-08-25:
Hi

Try create symbolyc link
cd /usr/share/pve-edk2-firmware
ln -s AAVMF_CODE.fd AAVMF_CODE.ms.fd

I had this error and fixed with this

Regards

yuhongwei380, 2023-08-25:
> Hi
> 
> Try create symbolyc link
> cd /usr/share/pve-edk2-firmware
> ln -s AAVMF_CODE.fd AAVMF_CODE.ms.fd
> 
> I had this error and fixed with this
> 
> Regards
> 

yeap，thank you，i remove the efi disk ，now my vm can be work！but this problem need to be fixed

jiangcuo, 2023-08-27:
New commit f0fdbd0411888fa76a5ae248798866a9e0658251   and new verison `3.20230228-4~bpo11+2` of pve-edk2-firmware has solved this issue.

closed 2023-08-27 by jiangcuo

yuhongwei380, 2023-08-28:
test OK ，My virtual machine  can start with efi-disk  and work well ！ Thank you！