jiangcuo/Proxmox-Arm64 #45: Getting ZFS Working

mikeotoole, 2023-09-05:
Hello, I'm looking to get ZFS working. This is on a Raspberry Pi 4. I generally followed the instructions here for a fresh install.
https://gist.github.com/gtx28/059ac9a0c85e722225c6d5150d659dec

Thanks for the help and thanks for getting Proxmox 8 working on the Pi!

Here is what I've tried.
```
$ sudo apt install zfs-dkms zfsutils-linux
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  zfs-zed
Suggested packages:
  debhelper nfs-kernel-server samba-common-bin zfs-initramfs
The following NEW packages will be installed:
  zfs-dkms zfs-zed zfsutils-linux
0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.
Need to get 2,841 kB of archives.
After this operation, 20.5 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://deb.debian.org/debian bookworm/contrib arm64 zfs-dkms all 2.1.11-1 [2,304 kB]
Get:2 https://global.mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 zfsutils-linux arm64 2.1.12-pve1 [473 kB]
Get:3 https://global.mirrors.apqa.cn/proxmox/debian/pve bookworm/port arm64 zfs-zed arm64 2.1.12-pve1 [63.8 kB]
Fetched 2,841 kB in 2s (1,166 kB/s)
Preconfiguring packages ...
Selecting previously unselected package zfs-dkms.
(Reading database ... 64528 files and directories currently installed.)
Preparing to unpack .../zfs-dkms_2.1.11-1_all.deb ...
Unpacking zfs-dkms (2.1.11-1) ...
Selecting previously unselected package zfsutils-linux.
Preparing to unpack .../zfsutils-linux_2.1.12-pve1_arm64.deb ...
Unpacking zfsutils-linux (2.1.12-pve1) ...
Selecting previously unselected package zfs-zed.
Preparing to unpack .../zfs-zed_2.1.12-pve1_arm64.deb ...
Unpacking zfs-zed (2.1.12-pve1) ...
Setting up zfsutils-linux (2.1.12-pve1) ...
zfs-import-scan.service is a disabled or a static unit, not starting it.
Setting up zfs-dkms (2.1.11-1) ...
Loading new zfs-2.1.11 DKMS files...
Building for 6.1.21-v8+
Module build for kernel 6.1.21-v8+ was skipped since the
kernel headers for this kernel do not seem to be installed.
Setting up zfs-zed (2.1.12-pve1) ...
Processing triggers for man-db (2.11.2-2) ...
Processing triggers for initramfs-tools (0.142) ...
$ sudo modprobe zfs
modprobe: FATAL: Module zfs not found in directory /lib/modules/6.1.21-v8+
$ sudo apt install dpkg-dev raspberrypi-kernel-headers linux-image-arm64
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package raspberrypi-kernel-headers
```

TheBossME, 2023-09-06:
generally it's easier to use btrfs on raspberry in fact of memory amount.

have you added the "contrib" to /etc/apt/sources.list ?

jiangcuo, 2023-09-06:
I don’t know where to download the header either. You can compile the kernel yourself or install a generic kernel with zfs integration.

```
#kernel repo.
deb https://global.mirrors.apqa.cn/proxmox/debian/kernel sid port
```

do

```
apt update && apt search pve-kernel-6.1*
apt install pve-kernel-6.1.X-generic pve-headers-xxxxxxxxxxxxx
```

kenji21, 2023-09-06:
To get kernel header, check the .deb download URL here: https://github.com/jiangcuo/Proxmox-Port/issues/2#issuecomment-1701581278

mikeotoole, 2023-09-06:
Thanks everyone for the options and details! I ended up doing the following and got it working.

```
sudo apt install dpkg-dev linux-image-arm64

wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel-headers_1.20230405-1_arm64.deb

sudo dpkg -i --force-all raspberrypi-kernel-headers_1.20230405-1_arm64.deb

sudo apt install zfs-dkms zfsutils-linux

sudo modprobe zfs
```

closed 2023-09-06 by mikeotoole