jiangcuo/Proxmox-Arm64 #46: How can i change the kernel with pve-kernel on nanopi t6?

SuperKali, 2023-09-12:
Hello,

I have a question actually i using a Linux 6.5.0-rc1-collabora-rockchip-rk3588 #1 SMP PREEMPT_DYNAMIC Fri Jul 28 15:49:06 UTC 2023
my question is how can i use pve-kernel https://github.com/jiangcuo/pve-port-kernel

Thank you in advice.

jiangcuo, 2023-09-13:
```bash
echo "deb https://mirrors.apqa.cn/proxmox/debian/kernel sid port" > /etc/apt/sources.list.d/pve-arm64-kernel.list
apt update
apt search pve-kernel
apt install pve-kernel-xxxx
```

SuperKali, 2023-09-13:
@jiangcuo i tried it, but after reboot doesn't load the new kernel

jiangcuo, 2023-09-14:
You should edit your uboot config.
The repo kernel is built for the general-purpose Arm64 PC

SuperKali, 2023-09-22:
@jiangcuo sorry for the reply late, but this kernel are compatible with nanopc t6, have a guide to make it work?

jiangcuo, 2023-09-28:
I'm sorry, I don't have the extra energy to compatible these boards because there are so many development boards for soc。If you're looking for ways to use the soc kernel, I recommend going to the community.

closed 2023-09-28 by SuperKali