jiangcuo/Proxmox-Arm64 #47: Does not install on Orange Pi 5

cantalupo555, 2023-09-27:
I'm using Orange Pi 5 with Armbian Debian, kernel 6.5 edge.

At the end of the command it displays an error:
`apt install proxmox-ve postfix open-iscsi`

I used the wiki as an installation reference:
https://github.com/jiangcuo/Proxmox-Arm64/wiki/Install-Proxmox-VE-on-Debian-bullseye

```
Creating config file /etc/perl/XML/SAX/ParserDetails.ini with new version
Setting up open-iscsi (2.1.3-5) ...
Created symlink /etc/systemd/system/sysinit.target.wants/iscsid.service → /lib/systemd/system/iscsid.service.
Created symlink /etc/systemd/system/iscsi.service → /lib/systemd/system/open-iscsi.service.
Created symlink /etc/systemd/system/sysinit.target.wants/open-iscsi.service → /lib/systemd/system/open-iscsi.service.
Setting up libwbclient0:arm64 (2:4.13.13+dfsg-1~deb11u5) ...
Setting up libjson-glib-1.0-0:arm64 (1.6.2-1) ...
Setting up libbytes-random-secure-perl (0.29-1) ...
Setting up libvirglrenderer1:arm64 (0.8.2-5+deb11u1) ...
Setting up proxmox-backup-client (2.4.1-1) ...
Setting up liblttng-ust-ctl4:arm64 (2.12.1-1) ...
Setting up libknet1:arm64 (1.24-pve2) ...
Setting up shared-mime-info (2.0-1) ...
Setting up python3-requests (2.25.1+dfsg-2) ...
Setting up swtpm (0.8.0~bpo11+3) ...
Setting up python3-ldb (2:2.2.3-2~deb11u2) ...
Setting up thin-provisioning-tools (0.9.0-1) ...
Setting up libqt5core5a:arm64 (5.15.2+dfsg-9) ...
Setting up libtemplate-perl (2.27-1+b3) ...
Setting up libxml-libxml-perl (2.0134+dfsg-2+b1) ...
update-perl-sax-parsers: Registering Perl SAX parser XML::LibXML::SAX::Parser with priority 50...
update-perl-sax-parsers: Registering Perl SAX parser XML::LibXML::SAX with priority 50...
update-perl-sax-parsers: Updating overall Perl SAX parser modules info file...
Replacing config file /etc/perl/XML/SAX/ParserDetails.ini with new version
Setting up libwww-robotrules-perl (6.02-1) ...
Setting up criu (3.16-2) ...
Setting up libqt5dbus5:arm64 (5.15.2+dfsg-9) ...
Setting up libhtml-parser-perl (3.75-1+b1) ...
Setting up libxslt1.1:arm64 (1.1.34-4+deb11u1) ...
Setting up libcrypt-ssleay-perl (0.73.06-1+b3) ...
Setting up libsndfile1:arm64 (1.0.31-2) ...
Setting up liblttng-ust0:arm64 (2.12.1-1) ...
Setting up libiscsi7:arm64 (1.19.0-3) ...
Setting up libgfrpc0:arm64 (9.2-1) ...
Setting up libavahi-client3:arm64 (0.8-5+deb11u2) ...
Setting up libio-socket-ssl-perl (2.069-1) ...
Setting up libhttp-message-perl (6.28-1) ...
Setting up gnutls-bin (3.7.1-5+deb11u3) ...
Setting up libhttp-negotiate-perl (6.01-1) ...
Setting up librados2 (17.2.6-pve1) ...
Setting up libhttp-cookies-perl (6.10-1) ...
Setting up libpulse0:arm64 (14.2-2) ...
Setting up libgfchangelog0:arm64 (9.2-1) ...
Setting up libhtml-tree-perl (5.07-2) ...
Setting up libspice-server1:arm64 (0.14.3-2.1) ...
Setting up xsltproc (1.1.34-4+deb11u1) ...
Setting up cifs-utils (2:6.11-3.1+deb11u1) ...
update-alternatives: using /usr/lib/aarch64-linux-gnu/cifs-utils/idmapwb.so to provide /etc/cifs-utils/idmap-plugin (idmap-plugin) in auto mode
Setting up libqt5network5:arm64 (5.15.2+dfsg-9) ...
Setting up libcephfs2 (17.2.6-pve1) ...
Setting up swtpm-tools (0.8.0~bpo11+3) ...
Adding group `tss' (GID 121) ...
Done.
Warning: The home dir /var/lib/tpm you specified can't be accessed: No such file or directory
Adding system user `tss' (UID 112) ...
Adding new user `tss' (UID 112) with group `tss' ...
Not creating home directory `/var/lib/tpm'.
Setting up libradosstriper1 (17.2.6-pve1) ...
Setting up lxc-pve (5.0.2-2) ...
Created symlink /etc/systemd/system/multi-user.target.wants/lxc-monitord.service → /lib/systemd/system/lxc-monitord.service.
Created symlink /etc/systemd/system/multi-user.target.wants/lxc-net.service → /lib/systemd/system/lxc-net.service.
[... 125 lines cut ...]
update-initramfs: Armbian: Converting to u-boot format: /boot/uInitrd-6.5.0-rc5-edge-rockchip-rk3588
Image Name:   uInitrd
Created:      Wed Sep 27 13:03:08 2023
Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
Data Size:    30158440 Bytes = 29451.60 KiB = 28.76 MiB
Load Address: 00000000
Entry Point:  00000000
update-initramfs: Armbian: Symlinking /boot/uInitrd-6.5.0-rc5-edge-rockchip-rk3588 to /boot/uInitrd
ln: failed to create symbolic link '/boot/uInitrd': Operation not permitted
update-initramfs: Symlink failed, moving /boot/uInitrd-6.5.0-rc5-edge-rockchip-rk3588 to /boot/uInitrd
renamed '/boot/uInitrd-6.5.0-rc5-edge-rockchip-rk3588' -> '/boot/uInitrd'
update-initramfs: Armbian: done.
Running hook script 'zz-proxmox-boot'..
Re-executing '/etc/kernel/postinst.d/zz-proxmox-boot' in new private mount namespace..
No /etc/kernel/proxmox-boot-uuids found, skipping ESP sync.
Processing triggers for libc-bin (2.31-13+deb11u6) ...
Processing triggers for rsyslog (8.2102.0-2+deb11u1) ...
Processing triggers for man-db (2.9.4-2) ...
Processing triggers for pve-ha-manager (3.6.1) ...
Errors were encountered while processing:
 pve-manager
 proxmox-ve
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@orangepi5:~# 
```

jiangcuo, 2023-09-28:
https://github.com/jiangcuo/Proxmox-Arm64/wiki/Install-Proxmox-VE-on-Debian-bullseye#troubleshooting

You should review your /etc/hosts file according to the instructions above.

cantalupo555, 2023-09-28:
This is my file /etc/hosts

```
127.0.0.1       localhost.localdomain localhost

# The following lines are desirable for IPv6 capable hosts
::1     localhost ip6-localhost ip6-loopback
ff02::1 ip6-allnodes
ff02::2 ip6-allrouters
2804:29b8:509d:e762:fe8a:f6f4:95c6:1400 debian
```

SuperKali, 2023-09-28:
@cantalupo555 you should add the ip address of the orange pi with the current hostname

Just like this:

[image]

cantalupo555, 2023-09-28:
> @cantalupo555 you should add the ip address of the orange pi with the current hostname
> 
> Just like this:
> 
> [image]

Hi, @SuperKali
I only have IPv6.
My IPv4 is local.

My hostname is orangepi5, so my file should be:
```
127.0.0.1      localhost
192.168.1.125 orangepi5

# The following lines are desirable for IPv6 capable hosts
::1     localhost ip6-localhost ip6-loopback
ff02::1 ip6-allnodes
ff02::2 ip6-allrouters
2804:29b8:509d:e762::68c orangepi5
```

Thanks you.

SuperKali, 2023-09-28:
@cantalupo555 if you see, when you do the command hostname --ip-address the ip address should be 192.168.1.125, this means you've configured it correctly.

cantalupo555, 2023-09-28:
@SuperKali

Thank you very much, it worked.
I'm using Armbian based on Debian 12 (bookworm).

```
root@orangepi5:~# hostname --ip-address
2804:29b8:509d:e762::7a3 192.168.1.113
```

I'm going to change the domain to orangepi5.cantalupo.com.br and open the port to have external access to the proxmox panel via IPv6.
[image]

[image]