jiangcuo/Proxmox-Arm64 #51: No Memory and SWAP Limits for CT's

Rappelkiste98, 2023-10-19:
I have multiple Debian Bookworm CT's running on a Raspberry PI 4B 8GB and they running fine,
but Memory and SWAP Settings from Proxmox are not working and all CT's have 8GB in "top".

[image]
[image]

Do you have a Fix or a Workaround for this Problem on Raspberry PI's?

TheBossME, 2023-10-20:
please modify cmdline.txt like 
nano /boot/cmdline.txt

put the following parameters to the end of the line

cgroup_enable=cpuset cgroup_enable=memory cgroup_memory=1

this will fix it and show memory properly

Regards
Bea

Rappelkiste98, 2023-10-23:
Thank you very much. The Problem is solved 👍

closed 2023-10-23 by Rappelkiste98