jiangcuo/Proxmox-Arm64 #52: Install proxmox 8 on Raspberry pi 4 with bookworm

Yngve0, 2023-10-22:
I try to follow the wiki to install on a raspberry pi 4 with Bookworm. Unfortunately is stopped with this:

W: Skipping acquire of configured file 'port/binary-armhf/Packages' as repository 'https://global.mirrors.apqa.cn/proxmox/debian/pve bookworm InRelease' does not seem to provide it (sources.list entry misspelt?)

jiangcuo, 2023-10-23:
Only for arm64.

closed 2023-10-23 by jiangcuo

TheBossME, 2023-10-23:
Change the /etc/apt/sources.list.d/pveport.list to:

deb [arch=arm64] https://global.mirrors.apqa.cn/proxmox/debian/pve bookworm port

In the current bookworm arm64lite image is armhf support not removed.

Try with this and let us know.
you can also install with the tested bookworm images from raspi.Debian.net.

Regards
Beatrice