jiangcuo/Proxmox-Arm64 #55: Network Issue

liberodark, 2023-11-05:
Hi,

Have create AlmaLinux CT can ping my PVE but have not internet access and i can't ping my gateway. 
You know if you have specific issue on arm with network ?

Also i cannot create debian CT have : 

`unable to open file '/etc/network/interfaces.tmp.2766' - No such file or directory`

Also have try to run cloudinit have network issue too.

Best Regards

TheBossME, 2023-11-05:
you are using a downstripped lxc image.

Start the lxc without dhcp configuration. Use dhclient to get ip and install the needed packages like ifupdown. (this is similar to Debian lxc arm64 images). 

It isn't an issue. This is related to your used lxc image.

After you've installed the missed packages, you can use it with dhcp configuration.

liberodark, 2023-11-05:
Hi,

DHCP or Static ins't work have only just 1 LXC with DHCP for test.
Also have install `ifupdown2` on PVE.

Best Regards

jiangcuo, 2023-11-06:
https://github.com/jiangcuo/Proxmox-Port/issues/3

TheBossME, 2023-11-06:
Is it clear now, that the creation of the lxc needs "NO NETWORK" configuration while in the LXC image is a directory missed /etc/network

1. do create a first LXC without any network modifications. This allow the proper creation. 

2. start the LXC Container and enter dhclient in the container

3. check if you have then an IP address received from your dhcp server.

4. do apt update and apt install ifupdown

5. stop the LXC and change the network setup to dhcp or a static ip.

6. restart LXC and check network connection. 

7. shutdown the LXC and make a backup and put this backup into your /var/lib/vz/template/cache directory for future installations, or do anytime the same steps while creating LXC containers. This is the actual work around on LXC's with missed /etc/network directories. 

hope this helps a little bit.
(tested with Debian bookworm images from the nightly build repository)

liberodark, 2023-11-06:
Hi, @TheBossME (Thank for your explain)

I think you speak for Debian 11 & 12, have understand your workaround.
But in Debian 10 have ifupdown installed :
`ifupdown/oldoldstable,now 0.8.35 arm64 [installed]` 
and /etc/network : 
```
ls /etc/network
if-down.d  if-post-down.d  if-pre-up.d  if-up.d  interfaces  interfaces.d
```
And on other LXC AlmaLinux 8 & 9 too.
But don't have network access.
That my issue.

Best Regards

TheBossME, 2023-11-06:
Check your Proxmox hypervisor network (disable vlan aware settings if possible) 

Looks like related to your hypervisor kernel. Didn't have that hardware. Try with backport kernels. 6.4.

liberodark, 2023-11-06:
Yes im using kernel by armbian for Odroid C4 im try to switch on kernel 6.4 edge.

closed 2023-11-06 by liberodark