jiangcuo/Proxmox-Arm64 #56: /usr/bin/dpkg returned an error code (1)

probablyQ, 2023-11-07:
I'm trying to figure out why linux-headers-arm64 isn't working.

i'm following this guide here: https://gist.github.com/gtx28/059ac9a0c85e722225c6d5150d659dec

i'm at line 115 and getting this return

Running the pre_build script:
*** Failed to find linux kernel source code at /usr/src/linux-6.1.21-v8+ - using default ***

Building module:
Cleaning build area...
make -j4 KERNELRELEASE=6.1.0-13-arm64 -C /lib/modules/6.1.0-13-arm64/build M=/var/lib/dkms/ceph-dkms/0.0.2/build.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.0-13-arm64 (aarch64)
Consult /var/lib/dkms/ceph-dkms/0.0.2/build/make.log for more information.
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
dkms: autoinstall for kernel: 6.1.0-13-arm64 failed!
run-parts: /etc/kernel/header_postinst.d/dkms exited with return code 11
Failed to process /etc/kernel/header_postinst.d at /var/lib/dpkg/info/linux-headers-6.1.0-13-arm64.postinst line 11.
dpkg: error processing package linux-headers-6.1.0-13-arm64 (--configure):
 installed linux-headers-6.1.0-13-arm64 package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of linux-headers-arm64:
 linux-headers-arm64 depends on linux-headers-6.1.0-13-arm64 (= 6.1.55-1); however:
  Package linux-headers-6.1.0-13-arm64 is not configured yet.

dpkg: error processing package linux-headers-arm64 (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 linux-headers-6.1.0-13-arm64
 linux-headers-arm64
E: Sub-process /usr/bin/dpkg returned an error code (1)

TheBossME, 2023-11-07:
Don't do that !!

For raspberry Pi it is not recommended to do upgrade processes from bullseye to bookworm. Read the documentations and recommendations for bookworm on raspi.

Without purging old packages from bullseye you will get failed as you described.

Thanks 🙏 

Use bookworm from the official bookworm build, and don't forget the [arch=arm64] parameter in the

/etc/apt/sources.list.d/pveport.list 

file 

or my personal opinion is using this without raspi related packages and without armhf architecture

https://raspi.debian.net/daily/raspi_4_bookworm.img.xz

probablyQ, 2023-11-07:
I had already commented out the pimox repo, or it was commented out for me maybe. I ran them apt update individually, which didn't have any problems before, anyway. I also went and added [arch=arm64] to the repos in /etc/apt/sources.list.d/pveport.list and /etc/apt/sources.list.d/pimox.list as was suggested. Then I ran apt dist-upgrade and it all worked out.

Thank you very much, I appreciate it.

closed 2023-11-07 by probablyQ