jiangcuo/Proxmox-Arm64 #58: pve-manager (8.0.8-1) Update failed

semool, 2023-11-23:
pve-manager (8.0.8-1) wird eingerichtet ...
Job for pvedaemon.service failed because the control process exited with error code.
See "systemctl status pvedaemon.service" and "journalctl -xeu pvedaemon.service" for details.
dpkg: Fehler beim Bearbeiten des Paketes pve-manager (--configure):
 »installiertes post-installation-Skript des Paketes pve-manager«-Unterprozess gab den Fehlerwert 1 zurück
Fehler traten auf beim Bearbeiten von:
 pve-manager
E: Sub-process /usr/bin/dpkg returned an error code (1)

systemctl status pvedaemon.service
● pvedaemon.service - PVE API Daemon
     Loaded: loaded (/lib/systemd/system/pvedaemon.service; enabled; preset: enabled)
     Active: activating (start) since Thu 2023-11-23 17:58:39 CET; 348ms ago
Cntrl PID: 3461791 (pvedaemon)
      Tasks: 1 (limit: 8754)
     Memory: 14.4M
     CGroup: /system.slice/pvedaemon.service
             └─3461791 "[pvedaemon]"

Nov 23 17:58:39 proxmox4 pvedaemon[3461791]: Compilation failed in require at /usr/share/perl5/PVE/JSONSchema.pm line 10.
Nov 23 17:58:39 proxmox4 pvedaemon[3461791]: BEGIN failed--compilation aborted at /usr/share/perl5/PVE/JSONSchema.pm line 10.
Nov 23 17:58:39 proxmox4 pvedaemon[3461791]: Compilation failed in require at /usr/share/perl5/PVE/INotify.pm line 20.
Nov 23 17:58:39 proxmox4 pvedaemon[3461791]: BEGIN failed--compilation aborted at /usr/share/perl5/PVE/INotify.pm line 20.
Nov 23 17:58:39 proxmox4 pvedaemon[3461791]: Compilation failed in require at /usr/share/perl5/PVE/Daemon.pm line 22.
Nov 23 17:58:39 proxmox4 pvedaemon[3461791]: BEGIN failed--compilation aborted at /usr/share/perl5/PVE/Daemon.pm line 22.
Nov 23 17:58:39 proxmox4 pvedaemon[3461791]: Compilation failed in require at /usr/share/perl5/PVE/Service/pvedaemon.pm line 7.
Nov 23 17:58:39 proxmox4 pvedaemon[3461791]: BEGIN failed--compilation aborted at /usr/share/perl5/PVE/Service/pvedaemon.pm line 7.
Nov 23 17:58:39 proxmox4 pvedaemon[3461791]: Compilation failed in require at /usr/bin/pvedaemon line 11.
Nov 23 17:58:39 proxmox4 pvedaemon[3461791]: BEGIN failed--compilation aborted at /usr/bin/pvedaemon line 11.

jiangcuo, 2023-11-23:
Did you edit "/etc/hosts" before upgrading?
if not
do 
```bash
apt download pve-manager=8.0.6
dpkg -i pve-manager_8.0.6_arm64.deb
```

anipad, 2023-11-23:
Same issue on my 3 machines with pve-manager (8.0.8-1), please see additional journalctl:
No editing of "/etc/hosts".

> journalctl -xeu pvedaemon.service
Nov 23 19:14:37 pimox3 pvedaemon[3188364]: Compilation failed in require at /usr/bin/pvedaemon line>
Nov 23 19:14:37 pimox3 pvedaemon[3188364]: BEGIN failed--compilation aborted at /usr/bin/pvedaemon >
Nov 23 19:14:37 pimox3 systemd[1]: pvedaemon.service: Control process exited, code=exited, status=2>
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ An ExecReload= process belonging to unit pvedaemon.service has exited.
░░
░░ The process' exit code is 'exited' and its exit status is 255.
Nov 23 19:14:37 pimox3 systemd[1]: Reload failed for pvedaemon.service - PVE API Daemon.
░░ Subject: A reload job for unit pvedaemon.service has finished
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A reload job for unit pvedaemon.service has finished.
░░
░░ The job identifier is 16463 and the job result is failed.

semool, 2023-11-23:
> Did you edit "/etc/hosts" before upgrading? if not do
> 
> ```shell
> apt download pve-manager=8.0.6
> dpkg -i pve-manager_8.0.6_arm64.deb
> ```

What should i edit in hosts file?
The Ip to hostname is set from the beginning like the readme file.
All was working fine before i upgraded today.
The manual download and install gives the same error: pvedaemon.service failed.

semool, 2023-11-23:
OK with try and error i found the Problem.
running "pvedaemon" in shell gives me:
Undefined subroutine &PVE::Syscall::SYS_mknod called at /usr/share/perl5/PVE/Syscall.pm line 11.
BEGIN failed--compilation aborted at /usr/share/perl5/PVE/Syscall.pm line 81.
Compilation failed in require at /usr/share/perl5/PVE/Tools.pm line 31.
BEGIN failed--compilation aborted at /usr/share/perl5/PVE/Tools.pm line 31.
Compilation failed in require at /usr/share/perl5/PVE/JSONSchema.pm line 10.
BEGIN failed--compilation aborted at /usr/share/perl5/PVE/JSONSchema.pm line 10.
Compilation failed in require at /usr/share/perl5/PVE/INotify.pm line 20.
BEGIN failed--compilation aborted at /usr/share/perl5/PVE/INotify.pm line 20.
Compilation failed in require at /usr/share/perl5/PVE/Daemon.pm line 22.
BEGIN failed--compilation aborted at /usr/share/perl5/PVE/Daemon.pm line 22.
Compilation failed in require at /usr/share/perl5/PVE/Service/pvedaemon.pm line 7.
BEGIN failed--compilation aborted at /usr/share/perl5/PVE/Service/pvedaemon.pm line 7.
Compilation failed in require at /usr/bin/pvedaemon line 11.
BEGIN failed--compilation aborted at /usr/bin/pvedaemon line 11.

So i edit "\usr\share\perl5\PVE\Syscall.pm" and remove line 19:
mknod => &SYS_mknod,

and then "usr\share\perl5\PVE\Tools.pm" and remove this part at line 1754:
sub mknod($$$) {
    my ($filename, $mode, $dev) = @_;
    return syscall(PVE::Syscall::SYS_mknod, $filename, int($mode), int($dev)) == 0;
}

After that make another apt dist-upgrade and pve-manager will configure OK an running fine.

anipad, 2023-11-23:
> OK with try and error i found the Problem. running "pvedaemon" in shell gives me: Undefined subroutine &PVE::Syscall::SYS_mknod called at /usr/share/perl5/PVE/Syscall.pm line 11. BEGIN [...]

Worked as described. Thank you.

semool, 2023-11-23:
Today Proxmox released 8.1.3.
Eventually with this version it will work again when we became a Update here.

jiangcuo, 2023-11-24:
Hello @semool @anipad ,
What is your `pveversion -v` and hardware ?

jiangcuo, 2023-11-24:
The bug is reproduced on `libpve-common-perl=8.0.10`

TheBossME, 2023-11-24:
It is now fixed (doing fresh install from repo)

Regards Beatrice

closed 2023-11-24 by semool