jiangcuo/Proxmox-Arm64 #65: Only a Info for RaspberryPi5 Kernel

semool, 2023-12-10:
When you run this Port on a Pi5 you should use the Kernel with 4K pagesize.
The Default Kernel for Pi5 use 16K and not all Software is compatible with this.
As a Example proxmox-backup-client, he cant download previous Manifest files and must upload all chunks everytime.
OK Short answer:
Add in your config.txt "kernel=kernel8.img" and reboot.

Now the RPi will boot with the 4K Kernel.

jiangcuo, 2023-12-10:
Create a wiki page ：https://github.com/jiangcuo/Proxmox-Port/wiki/RaspberryPi

This page will collect questions about Raspberry Pi

Thanks for sharing.

zapo20, 2023-12-13:
Hello,

[https://www.makeuseof.com/raspberry-pi- ... o-install/](https://www.makeuseof.com/raspberry-pi-proxmox-virtualization-how-to-install/)
I am installing Proxmox by installing the Raspi OS 64 bit Lite version on the 128GB SD Card on the Raspi 5 listed in the URL.
Even though I did everything correctly step by step, https://ipaddress:8006/ Proxmox GUI does not open.

I tried many times. There are many posts about Raspi 4 on the internet. The URL above is similar and in addition it says appropriate in 4 and 5.
I made various attempts. I am no stranger to my job, but for some reason the Proxmox interface does not open.

My goal is to establish Home Assistant by virtualizing on Raspi 5 and to create a lab environment for myself with Home Assistant style Appliances with the remaining hardware.

I am sharing the logs from my last installation attempt for Proxmox.
I would appreciate your help.
(If you have a different suggestion regarding virtualization, I would consider it as well)

`=========================================================================================
 Begin installation, Normal duration on a default RPi4 ~ 30 minutes, be patient...
=========================================================================================

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
100  3143  100  3143    0     0  11297      0 --:--:-- --:--:-- --:--:-- 11265
OK
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease                                        
Hit:4 http://archive.raspberrypi.com/debian bookworm InRelease                                       
Get:5 https://raw.githubusercontent.com/pimox/pimox7/master dev/ InRelease [2186 B]
Get:6 https://raw.githubusercontent.com/pimox/pimox7/master dev/ Packages [101 kB]
Fetched 103 kB in 1s (106 kB/s)  
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
W: https://raw.githubusercontent.com/pimox/pimox7/master/dev/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package 'dhcpcd5' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ceph-fuse : Depends: libfmt7 (>= 7.1.3+ds1) but it is not installable
 libpve-rs-perl : Depends: perlapi-5.32.1 but it is not installable
 libpve-u2f-server-perl : Depends: perlapi-5.32.1 but it is not installable
 librados2-perl : Depends: perlapi-5.32.1 but it is not installable
 lxc-pve : Depends: libgnutlsxx28 but it is not installable
 pve-cluster : Depends: perlapi-5.32.1 but it is not installable
 pve-qemu-kvm : Depends: liburing1 (>= 0.7) but it is not installable
E: Unable to correct problems, you have held broken packages.
Reading package lists... Done
Building dependency tree... Done
[... 19 lines cut ...]
=========================================================================================
 ! FIXING NETWORK CONFIGURATION.... ERRORS ARE NOMALAY FINE AND RESOLVED AFTER REBOOT ! 
=========================================================================================

cp: cannot stat '/usr/share/javascript/proxmox-widget-toolkit/proxmoxlib.js': No such file or directory
sed: can't read /usr/share/javascript/proxmox-widget-toolkit/proxmoxlib.js: No such file or directory

=========================================================================================
                        ! INSTALATION COMPLETED ! WAIT ! REBOOT ! 
=========================================================================================

    after reboot the PVE web interface will be reachable here :
      --->   https://10.10.2.18:8006/  <---
      
         run --->  apt upgrade -y  <---
           in a root shell to complete the installation.
           

Broadcast message from root@SarpinarRasPi5 on pts/2 (Thu 2023-12-07 22:56:54 +03):

The system will reboot now!

Broadcast message from root@SarpinarRasPi5 on pts/2 (Thu 2023-12-07 22:56:54 +03):

The system will reboot now!`

TheBossME, 2023-12-13:
Bullseye is pimox 7 (proxmox 7.x release)
Bookworm needs Proxmox 8 installation and the different repo.

Install it in proper way and not pimox 7 in bookworm.
This goin failed.

zapo20, 2023-12-13:
@TheBossME 

Is there a document with installation steps for pimox 7's bookworm? I also have the chance to download it from the correct address. Can you help me?

TheBossME, 2023-12-13:
Have you access to jiung's discord server ?

zapo20, 2023-12-13:
@TheBossME 

Thank you sir for your return.

Unfortunately, I do not have access to jiung's Discord server.

Frankly, I understood your explanations. However, if there is a location where the steps you describe are written, can you share its address?

I've been working hard and I'm short on time :(

I don't want to tire you either.

zapo20, 2023-12-13:
> Jiung'un discord sunucusuna erişimin var mı?
> 
> Raspberry pi modellerine özel adımlarını anlattım.
> 
> RPi genel olarak u-Boot'u kullanır. Pi'de UEFI önyükleyici ile Başlatma yapıyorsanız. .iso'yu kullanamazsınız
> 
> Disk kurulumundan sonra UEFI ekranlarının efi bölümünün kopyalanması gerekir. Daha sonra Raspi uefi uygulamalarını kullanır.
> 
> Raspi 5'in aslında uefi uygulamalarını destekleyip desteklemediğini bildiğini ve ayrıca pi 5 için uefi önyükleyici hizmetlerini de aldığını biliyorum.

Can you help me? I'm sorry.

semool, 2023-12-13:
@zapo20 
This is not a discussion Forum.
Please stop spaming.
This Issue here has nothing todo with your Problems.

closed 2023-12-13 by semool