jiangcuo/Proxmox-Arm64 #66: LXC containers stopped working on PVE 8.1

SuperKali, 2023-12-10:
Hello,

I am trying to run an LXC instance on my NanoPC T6, but I am getting this error:

```bash
Formatting '/var/lib/vz/images/103/vm-103-disk-0.raw', fmt=raw size=10737418240 preallocation=off
Creating filesystem with 2621440 4k blocks and 655360 inodes
Filesystem UUID: e04a8139-c0be-4142-85ae-f1361adbeddd
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632
extracting archive '/var/lib/vz/template/cache/debian-bookworm-20231124_arm64.tar.xz'
Total bytes read: 452454400 (432MiB, 53MiB/s)
Detected container architecture: arm64
unable to open file '/etc/network/interfaces.tmp.1541833' - No such file or directory
TASK ERROR: unable to create CT 103 - error in setup task PVE::LXC::Setup::post_create_hook
```

In the previous version it worked, now I don't understand if I'm missing something or if some step needs to be done to make it works.

Regards.

jiangcuo, 2023-12-11:
https://github.com/jiangcuo/Proxmox-Port/issues/3

SuperKali, 2023-12-11:
The problem is only debian.

closed 2023-12-11 by SuperKali

TheBossME, 2023-12-11:
it's easy to fix.

1. create a machine with actual Debian 12 template without changing network settings (leave as static).

2. start the container 

3. run dhclient

4. apt update & apt upgrade

5. install ifupdown package 

Reboot LXC Container 

6. make a backup (dump directory)

7. copy and rename or move your ...tar.zst file to the /var/lib/template/cache directory.

Now you are able to push new containers without network setup issue.