jiangcuo/Proxmox-Arm64 #68: Debian 12 LXC with DHCP Fails to Boot

soxrok2212, 2023-12-15:
Hi,

When I set up a fresh Debian 12 LXC container, it fails to boot when I select DHCP in my ethernet interface:
```
run_buffer: 322 Script exited with status 1
lxc_init: 844 Failed to run lxc.hook.pre-start for container "138"
__lxc_start: 2027 Failed to initialize container "138"
TASK ERROR: startup for container '138' failed
```

Setting the interface to static works, but of course it won't automatically get an IP address.

Here's the config:
```
arch: arm64
cores: 1
features: nesting=1
hostname: debian12
memory: 1024
net0: name=eth0,bridge=vmbr0,firewall=1,hwaddr=XX:XX:XX:XX:XX:XX,ip=dhcp,tag=10,type=veth
ostype: debian
rootfs: local:138/vm-138-disk-0.raw,size=16G
swap: 512
unprivileged: 1
```

TheBossME, 2023-12-15:
Debian specific problem, 

Create /etc/network folder

Set to dhcp 

Reboot

Dhclient

Apt update

Apt install ifupdown

Should fix it.

soxrok2212, 2023-12-15:
> Debian specific problem,
> 
> Create /etc/network folder
> 
> Set to dhcp
> 
> Reboot
> 
> Dhclient
> 
> Apt update
> 
> Apt install ifupdown
> 
> Should fix it.

This worked!

closed 2023-12-15 by soxrok2212